wolfSSL/wolfTPM #65: No output with examples and benchmark

SherlyFU, 2019-03-12:
Hello, wolfSSL team.

I am using wolfTPM project for my thesis research. Unfortunately, I could not have any valid output after launching examples and benchmark files. The following is the output of running bench:
```bash
pi@raspi-fx:~/Software/wolfTPM/examples/bench $ ./bench
TPM2 Benchmark using Wrapper API's
```
This is the output of running wrap test:
```bash
pi@raspi-fx:~/Software/wolfTPM $ ./examples/wrap/wrap_test
TPM2 Demo for Wrapper API's
```
I am using a Raspberry Pi 3, with Infineon SLB9670 attached. The OS is the latest stable version of Raspbian. I followed exactly the official wolfTPM tutorial, a.k.a README.md of this project, with the same commands. I tried two combinations of different versions: wolfTPM 1.4 with wolfSSL 3.15.7, and wolfTPM master branch with wolfSSL master branch. I enabled my TPM module with this instruction [https://letstrust.de/archives/19-Success!.html](url).

Thank you in advance and have a nice day.

Xiao.

dgarske, 2019-03-12:
Hi SherlyFU (Xiao),

That usually means the library was unable to communicate with the TPM module. 

You can debug by building using `./configure --enable-debug=verbose`.

The default configure enables support for the SLB9670 on `/dev/spidev0.1`.

First make sure you have the `/dev/spidev0.1` device available. If not make sure SPI is enabled by:
1. `sudo vim /boot/config.txt`
2. Uncomment `dtparam=spi=on`

You should make sure that is the correct chip select for your hardware (could also be `/dev/spidev0.0` if not using a standard board). You might also try reducing the SPI clock speed just in case that's an issue. Those changes would be made here:
https://github.com/wolfSSL/wolfTPM/blob/master/examples/tpm_io.c#L84

If you continue to have issues please send the logs and output from `dmesg | grep spi`.
Thanks,
David Garske, wolfSSL

dgarske, 2019-03-12:
Hi SherlyFU (Xiao),

Additionally the wolfTPM library only supports the TPM 2.0 specification. The SLB9670 must have TPM 2.0 firmware loaded on it.

Thanks,
David Garske, wolfSSL

SherlyFU, 2019-03-12:
Hi dgarske,

Thank you for your rapid reply! Actually, I enabled SPI by using exactly the same method. However, my TPM module is loaded as "/dev/tpm0" instead of "/dev/spidev0.1". I am pretty sure that I am using TPM 2.0 version of SLB 9670, with corresponding firmware included. Personally, I believe this issue has something to do with different naming of TPM module. Is there any clue for renaming this device?
For better understanding of this issue, I post the debug output of bench.
```bash
pi@raspi-fx:~/Software/wolfTPM/examples/bench $ ./bench
TPM2 Benchmark using Wrapper API's
TPM2_Init failed 257: TPM_RC_FAILURE: Commands not being accepted because of a TPM failure
```
Cheers,
SherlyFU

dgarske, 2019-03-12:
Hi SherlyFU (Xiao),

The wolfTPM library has its own TIS interface to communicate directly to the hardware via SPI. The Infineon TPM_TIS_SPI `/dev/tpm0` driver takes over (sits on top of) the `/dev/spidev0.1` interface. To use wolfTPM you'll have to disable the Infineon TIS driver. I believe you can comment out the `dtoverlay=tpm-slb9670` or `dtoverlay=letstrust-tpm` option in your `/boot/config.txt` and reboot. If that doesn't work you may need to run `menuconfig` and disable the TPM driver module. If you find an easier way to do this please let me know and I can update the README.md and user manual with these details.

Thanks,
David Garske, wolfSSL

SherlyFU, 2019-03-12:
Hi dgarske (David Garske),

Thank you very much for your instant reply! Your suggestion really helps me out of a big trouble. I will start reading official manuals carefully. Case closed!

Cheers,
SherlyFU (Xiao)

closed 2019-03-12 by SherlyFU