wolfSSL/wolfTPM #98: Can't read from NV memory

horver, 2020-05-28:
Hello,

I want to store 32-bytes of data in TPM's NV memory, then read it back (using an SLB9670). I used your example in wrap_test.c with success. After refactoring my code, I got an error when reading from the NV memory. I tried to restore the changes, and I tried the example code too, but couldn't get it to work. 

Testing code:
```
void test(WOLFTPM2_DEV *dev)
{
    int rc = -1;
    WOLFTPM2_HANDLE parent;
    WOLFTPM2_NV nv;
    word32 nvAttributes = 0;
    WOLFTPM2_BUFFER message, plain;

    XMEMSET(&parent, 0, sizeof(parent));
    parent.hndl = TPM_RH_OWNER;

    rc = wolfTPM2_GetNvAttributesTemplate(parent.hndl, &nvAttributes);
    if (rc != 0) return;

    rc = wolfTPM2_NVCreateAuth(dev, &parent, &nv, 0x01800000,
        nvAttributes, 32, (byte*)gNvAuth, sizeof(gNvAuth)-1);
    if (rc != 0 && rc != TPM_RC_NV_DEFINED) return;
   
   if (rc != TPM_RC_NV_DEFINED) {
        message.size = 32; 
        XMEMSET(message.buffer, 0x42, message.size);
        rc = wolfTPM2_NVWriteAuth(dev, &nv, 0x01800000,
            message.buffer, message.size, 0);
        if (rc != 0) return;
    }

    plain.size = 32;
    rc = wolfTPM2_NVReadAuth(dev, &nv, 0x01800000,
        plain.buffer, (word32*)&plain.size, 0);
    if (rc != 0) return;
}
```

The output it produces:
```
TPM2_NV_DefineSpace failed 332: TPM_RC_NV_DEFINED: NV Index or persistent object already defined
TPM2_NV_Read failed 388: TPM_RC_VALUE: Value is out of range or is not correct for the context
```

However, after clearing the chip, its seems to work once:
```
TPM2_NV_DefineSpace: Auth 0x40000001, Idx 0x1800000, Attribs 0x33947654, Size 64
TPM2_NV_Write: Auth 0x1800000, Idx 0x1800000, Offset 0, Size 32
TPM2_NV_Read: Auth 0x1800000, Idx 0x1800000, Offset 0, Size 32
```
But after I ran my test code again, it will still produces the TPM_RC_VALUE error.

I compiled wolfTPM with v1.7;v1.8 and the latest commit.

What could be the problem?

dgarske, 2020-05-28:
Hi @horver ,

This is because the auth is not set in the case when `wolfTPM2_NVCreateAuth` is called and the handle already exists.

The right solution would be to call `wolfTPM2_NVReadPublic` first to see if it exists. If it does then you'll need to set the NV handle like this:

```
nv.handle.hndl = (TPM_HANDLE) 0x01800000;
nv.handle.auth.size = sizeof(gNvAuth) - 1;
XMEMCPY(nv.handle.auth.buffer, gNvAuth, nvhandle.auth.size);
```

Then call `wolfTPM2_NVReadAuth`.

You do bring up a good improvement that could be made to the wrapper, so I've pushed a PR here:
https://github.com/wolfSSL/wolfTPM/pull/99

Thanks,
David Garske, wolfSSL

tomoveu, 2020-05-28:
Hi @horver,

This also means, you should handle the return code `TPM_RC_NV_DEFINED` in your application not as error case. After #99 the wrapper now sets the auth and you can perform the NVReadAuth.

Of course, you are left with the challenge to know that you are the owner of this NV field. This is why you should check the outcome of NV_ReadAuth to figure if the NV field written matches your auth, thus verifying "aha , this is my field and the value inside is the one i stored" :)

ps: @dgarske nice fix! Simple and quick.

closed 2020-05-29 by embhorn

horver, 2020-05-29:
Hello @tomoveu @dgarske,

Thank you very much for your answer and the quick fix!
Now I understand the mechanism and it works! :)

PR #99 (2020-05-29): If creating an NV and it already exists, set auth and handle anyways
Fixes #98
commit 2359d1b5 (2020-05-29): If creating an NV and it already exists, set auth and handle anyways. Fixes #98.