wolfSSL/wolfTPM #106: Quote example (PCR attestation)

tomoveu, 2020-07-15:
Making a Quote example using wolfTPM, my structure currently looks like this

./example/pcr
- quote  (attestation of PCR values / signing PCRs in a TPM2.0 Quote)
- extend (have a generaic wolfTPM tool to extend PCRs with arbitrary values, so the quotes can differ)

I think I can have this ready by EoW and I will try to push something before the weekend for review.

@dgarske  are you OK with one folder for more than one example, as long as they are on the same tech topic(PCR)?

ps: initially, I named the folder attestation, but then people may confuse with the timestamp folder which is also form of attestation, although by TCG spec attestation is signing PCRs and GetTime is just another Attestation operation AFAIK.

dgarske, 2020-07-15:
Hi @tomoveu : Yes putting multiple PCR examples into the same folder is fine. Your plan sounds great and I like the `pcr` folder name.

tomoveu, 2020-07-23:
Now that #107 is merged, we might think about a verifier of the generated TPM2 Quote. Essentially, verifying the signature.

tomoveu, 2021-03-18:
We have examples for:

1. PCR reset
2. PCR extend
3. Quote
4. Quote already stores the output into a file
* We are only missing a signature verifier. 

@dgarske I think the verifier example could be one for both Quote and GetTime #105 

Because both TPM 2.0 commands produce TPM2B_ATTEST and TPMT_SIGNATURE as their output. The latter holds the signature. In that case, we might be looking at a total of two days effort to have a verifier that supports both Quote and GetTime.

tomoveu, 2021-03-18:
Remaining enhancements will be address by #153

closed 2021-03-18 by tomoveu