wolfSSL/wolfTPM #110: Add TPM 2.0 Parameter Encryption support

dgarske, 2020-07-24:
XOR then AES CFB
@tomoveu

tomoveu, 2020-07-31:
Quick update

Had a "fun" tiny bug on the XOR symmetric attributes that made me sweat. Once solved, I figured out how to use unbounded, unsalted session for XOR parameter encryption. Turns out the logical first choice is the one with most specifics....

TCG Spec, Part 1

Chapter 21.2 
> the XOR obfuscation is (parameter, hashAlg, **sessionValue**, nonceNewer, nonceOlder)

Chapter 21.1
> If the session is not being used for authorization, **sessionValue is sessionKey.**

Chapter 19.6.8
> **the sessionKey is Empty buffer** when tpmKey and bind are TPM_RH_NULL (i.e unbounded, unsalted session).

So, if we have an empty buffer and no shared secret, how do we have parameter encryption? Well, see below ...

Chapter 19.6.9
> A session started using this format (unbound, unsalted) can be used for parameter encryption while executing TPM
commands. However, during these commands, **the key used to encrypt the parameter will only use the
authValue** of the object being accessed by the commands in the key generation, _so the strength of the
encryption will be no better than the entropy in the authValue of the object_.

Back to Chapter 19.6.8 we find 
`sessionKey ≔ KDFa(sessionAlg, (authValue || salt), “ATH” , nonceTPM, nonceCaller, bits)`
 and later clarifies
> authValue if bind is not TPM_RH_NULL,
> **a TPM2B_AUTH.buffer** that is found in the sensitive area of a TPM entity;
> otherwise, an Empty Buffer

Also, the tradeoff now becomes obvious, but funny it is briefly mentioned on one single place. This is the result of extensive digging into the TCG Spec until I find the answers. Unbounded & unsalted session is supposed to be the simplest one, yet it is also the one with most specifics to get parameter encryption(surprisingly).

Anyways, both big challenges are solved - TPM session and SessionKey. I just need to use the KDFa and replace the parameter with the encrypted buffer. I will use your KDFa port @dgarske with Woflcrypt. Hopefully, no new surprises arise and next week I am done with the XOR. And I move to the CFB/AES soon after :)

Cheers,
Dimi

dgarske, 2020-09-30:
Support for XOR parameter encryption was merged via #PR #122. AES CFB mode and keygen example are next.

tomoveu, 2020-10-09:
Solving the parameter encryption issue found #129 before continuing with the addition of AES CFB

I would expect #129 is solved soon and we would have PR for AES CFB next week.

tomoveu, 2020-12-03:
Thanks to great teamwork, full parameter encryption support is now added to wolfTPM by #129.

Kudos @dgarske :100: Very hard work at the end, with deep core redesign of wolfTPM to support HMAC sessions and full parameter encryption.

Looking forward to wolfTPM's major release :1st_place_medal:

closed 2020-12-03 by tomoveu