wolfSSL/wolfTPM #150: I2C driver for Nuvoton NPCT75x TPM2.0 module

uart-aplex, 2021-03-12:
Hi,
Can I use the Nuvoton NPCT75x TPM2.0 module with I2c interface? Where can I get the I2C driver software if it can be used? I wrote to ask Nuvoton, but Nuvoton said that they do not have a driver of 75x for LINUX yet.

tomoveu, 2021-03-12:
Hi @uart-aplex , we support the NPCT750 and we support I2C interface for TPM 2.0 modules.

Typically, for a baremetal system we have the` --enable-i2c` configuration switch. Have you tried any of our examples?

Could you please give details about your platform:
- SoC or MCU , board
- OS/RTOS/baremetal
-  and how you have connect your TPM 2.0 module

Also, it might be easier to send us detailed information at support@wolfssl.com

FreyrChen, 2021-03-12:
Dear Sir：
       1.  I currently use it on RK3399, the platform is ARM64.
       2.  Os si  Linux + Buildroot 。 Kernel Version is 4.4
       3. I have successfully compiled wolfTPM into ARM64, and wolfssl has also been compiled successfully.
       4. The configure script used for compilation is as follows:
    ```
           ./configure  \
    --prefix=/home/aplex/wolfTPM/out \
    --enable-nuvoton CC=aarch64-linux-gnu-gcc   \
    --host=aarch64 --libdir=/home/aplex/wolfTPM/out/lib  \
    --includedir=/home/aplex/wolfTPM/out/include \
    --enable-devtpm \
    --enable-i2c
    ```
     5. I put the compiled example directory into our product and run it, and the following error is displayed:
     ```
[root@SBC_7832:/wolfTPM]# ./examples/bench/bench
TPM2 Benchmark using Wrapper API's
        Use Parameter Encryption: NULL
Loading SRK: Storage failed 0x101: TPM_RC_FAILURE
Failure 0x101: TPM_RC_FAILURE
     ```
     6. We currently suspect that the lack of NPCT750 Linux driver caused this result.

tomoveu, 2021-03-12:
@FreyrChen thank you for this information

`--enable-devtpm` is for native Linux support, using /dev/tpm0 
`--enable-i2c` is for baremetal I2C support, for example STM32F4

These two switches are not used together. You have to use only `--enable-i2c` and recompile. 

Then, check your IO Callback. WolfTPM uses a single function for IO Callback. In our `exampes/tpm_io.c` you could find an example IO Callback for STM33 that use I2C interface 1 and has TPM2_IC_ADDR of 0x2E. This is the default I2C address defined by the Trusted Computing Group(TCG)).

```
   #ifdef WOLFTPM_I2C
        /* I2C - (Only tested with ST33HTPH I2C) */
        #define TPM2_I2C_ADDR 0x2e
        #define TPM2_I2C_DEV  "/dev/i2c-1"
        #define TPM2_I2C_HZ   400000 /* 400kHz */
```

Make sure your TPM is connected to the correct I2C interface and the correct I2C_ADDR.

It may that you have connected your TPM to `/dev/i2c-2 `, for example.

Pleaset let me know if this helps.

tomoveu, 2021-03-12:
@FreyrChen  I had a typo in the previous response

`--enable-i2c` is for Linux `/dev/i2c-1` or baremetal I2C support (like STM32F4 I2C).

tomoveu, 2021-03-15:
@FreyrChen @uart-aplex 

You could try https://github.com/Nuvoton-Israel/tpm_i2c_ptp for Linux kernel 5.2

Or you could use wolfTPM with any Linux kernel version, including your 4.4, including latest 5.11

tomoveu, 2021-03-23:
Hi @FreyrChen  @uart-aplex we would really like to find out more about your project progress. Please contact us at support@wolfssl.com , so we can help more.

tomoveu, 2021-04-08:
Hi @FreyrChen @uart-aplex I see no further questions, so I am going to close this issue.

Please contact us at support@wolfssl.com if you need more help.

Thanks,
Dimi

closed 2021-04-08 by tomoveu