wolfSSL/wolfTPM #188: Can't clear TPM chip

horver, 2021-09-25:
Hello,

I would like to ask a question about clearing the SLB9670 TPM chip.
After updating wolfTPM to v2.2.0, I had changed the handle authorization password and then I wanted to reset the chip using `wolfTPM2_Clear`. However the clear fails with:
```
TPM2_Clear failed 2337: TPM_RC_EXCLUSIVE: Command failed because audit sequence required exclusivity
```
Unfortunately, I didn't found any information about this error.

Can you please help, what commands can I use to fix the reset?

dgarske, 2021-09-27:
Hi @horver ,

Are you using the /dev/spidev interface, /dev/tpm0, Windows TBS or something else to communicate with the TPM? The TPM2_Clear is only allowed when using the locality 0, which is our default when using the internal TIS layer and /dev/spidev. Do other TPM command was fine? Perhaps this TPM has a policy auth set to prevent this without authenticating. Have you reviewed the TPM 2.0 specification around the TPM2_Clear command?

Thanks,
David Garske, wolfSSL

dgarske, 2022-07-15:
Hi @horver ,

Some time has passed with no update, so I wanted to check in. Did you find a resolution to this issue?

Note it is possible to disable the `TPM2_Clear` with the `TPM2_ClearControl`.

Thanks,
David Garske, wolfSSL

closed 2023-01-30 by dgarske