wolfSSL/wolfTPM #222: Error handling in C# wrapper

zhichao-h, 2022-07-06:
The wolftpm.cs bridges the native calls and transparently returns the error codes. This forces the high level code must research the different error returns. This can be hard to use or abuse.

I recommend raise C# Exceptions such as WolfTpmException("message", error_code) when the return code indicates error.
So the following code:
```
[DllImport(DLLNAME, EntryPoint = "wolfTPM2_GetRandom")]
        private static extern int wolfTPM2_GetRandom(IntPtr dev,
                                                     byte[] buf,
                                                     int len);
        public int GetRandom(byte[] buf)
        {
            return wolfTPM2_GetRandom(device, buf, buf.Length);
        }
``` 
should better be implemented as
```
[DllImport(DLLNAME, EntryPoint = "wolfTPM2_GetRandom")]
        private static extern int wolfTPM2_GetRandom(IntPtr dev,
                                                     byte[] buf,
                                                     int len);
        public void GetRandom(byte[] buf)
        {
            int ret = wolfTPM2_GetRandom(device, buf, buf.Length);
            if ( ret != TPM_RC_SUCCESS)
            {
                 throw new wolfTpm2Exception("wolfTPM2_GetRandom failed", ret);
            }
        }
```

With this modification, the high level code will not need to get into the details of understanding the error codes from native side.

dgarske, 2022-07-06:
Hi @zhichao-h , also great suggestion!

Does this look like a good example for what `WolfTpm2Exception` would look like?

```
[Serializable]
public class WolfTpm2Exception : Exception
{
    public int ErrorCode { get; }

    public WolfTpm2Exception() { }

    public WolfTpm2Exception(string message)
        : base(message) { }

    public WolfTpm2Exception(string message, Exception inner)
        : base(message, inner) { }

    public StudentNotFoundException(string message, int errorCode)
        : this(message)
    {
        ErrorCode = errorCode;
    }
}
```

Thanks,
David Garske, wolfSSL

dgarske, 2022-07-15:
Fixed in PR #224. Marking closed.

closed 2022-07-15 by dgarske