wolfSSL/wolfTPM #278: Building in MSYS2: undefined reference to `__imp_wolfSSL.._

ETHwsch, 2023-07-10:
I followed the instructions to build wolfSSL and wolfTMP as described in the manual:

export PREFIX=$PWD/tmp_install
cd wolfssl
./autogen.sh
./configure --prefix="$PREFIX" --enable-wolftpm
make

After calling make I am getting undefined references. Any ideas? Thanks for a feedback.

root1@.... UCRT64 ~/wolfssl
$ make
make -j7  all-recursive
make[1]: Entering directory '/home/root1/wolfssl'
make[2]: Entering directory '/home/root1/wolfssl'
make[2]: warning: -j7 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/benchmark/benchmark.o
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
 ..
 ..
  CC       tests/unit_test-w64wrapper.o
  CC       tests/unit_test-srp.o
  CC       tests/unit_test-quic.o
  CC       examples/client/tests_unit_test-client.o
  CC       examples/server/tests_unit_test-server.o
  CCLD     src/libwolfssl.la

*** Warning: linker path does not have real file for library -lws2_32.
*** I have the capability to make that library automatically link in when
*** you link to this library.  But I can only do this if you have a
*** shared version of the library, which you do not appear to have
*** because I did check the linker path looking for a file starting
*** with libws2_32 but no candidates were found. (...for file magic test)

*** Warning: linker path does not have real file for library -lcrypt32.
*** I have the capability to make that library automatically link in when
*** you link to this library.  But I can only do this if you have a
*** shared version of the library, which you do not appear to have
*** because I did check the linker path looking for a file starting
*** with libcrypt32 but no candidates were found. (...for file magic test)
*** The inter-library dependencies that have been dropped here will be
*** automatically added whenever a program is linked with this library
*** or is declared to -dlopen it.

*** Since this library must not contain undefined symbols,
*** because either the platform does not support them or
*** it was explicitly requested with -no-undefined,
*** libtool will only create a static version of it.
  CCLD     wolfcrypt/benchmark/benchmark.exe
  CCLD     wolfcrypt/test/testwolfcrypt.exe
  CCLD     examples/benchmark/tls_bench.exe
  CCLD     examples/echoclient/echoclient.exe
  CCLD     examples/client/client.exe
  CCLD     examples/echoserver/echoserver.exe
  CCLD     examples/server/server.exe
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: wolfcrypt/test/test.o:test.c:(.text+0xf9): undefined reference to `__imp_wc_GetErrorString'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: wolfcrypt/test/test.o:test.c:(.text+0x285): undefined reference to `__imp_wc_InitSha3_224'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: wolfcrypt/test/test.o:test.c:(.text+0x296): undefined reference to `__imp_wc_Sha3_224_Update'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: wolfcrypt/test/test.o:test.c:(.text+0x29d): undefined reference to `__imp_wc_Sha3_224_GetHash'
....

dgarske, 2023-07-10:
Hi @ETHwsch ,

I tried reproducing this with the latest wolfSSL and MSYS2 (x86_64-pc-msys), but was unable to. Can you tell me more about your platform (Windows/Linux), compiler and wolfSSL version? Is it possible your system is finding a conflicting wolfSSL library or options.h?

You might consider disabling the examples and tests `--disable-examples --disable-crypttests`.

Please try with `./configure --enable-wolftpm --disable-examples --disable-crypttests`.

Thanks,
David Garske, wolfSSL.

dgarske, 2023-07-10:
Hi @ETHwsch ,

For wolfTPM configure try using: `./configure --with-wolfcrypt="$PREFIX"`. This will setup the path for the wolfcrypt lib/include. Note: I just discovered that having a space in the path isn't handled correctly, so make sure your "PREFIX" path does not contain spaces.

If you get any other errors with configure please share the config.log.

Thanks,
David Garske, wolfSSL

ETHwsch, 2023-07-10:
I am getting:

```
./configure --with-wolfcrypt="$PREFIX" --enable-winapi
....
checking whether more special flags are required for pthreads... no
checking for PTHREAD_PRIO_INHERIT... no
checking for wolfSSL/wolfCrypt
... configure: prefix /ucrt64
configure: error: wolfCrypt path error (/home/root1/tmp_install): missing lib and include

```
Attached the log
[config.log](https://github.com/wolfSSL/wolfTPM/files/12006405/config.log)

dgarske, 2023-07-10:
Hi @ETHwsch ,

Sorry I assumed you had installed wolfSSL into that environment variable path. Does `/home/root1/tmp_install` have a `lib` and `include` directory from wolfSSL `make install`?

```
configure:15381: checking for wolfSSL/wolfCrypt

configure:15391: prefix /ucrt64
configure:15403: error: wolfCrypt path error (/home/root1/tmp_install): missing lib and include
```

Here are the full builds steps for MSYS2:

```
git clone --depth=1 https://github.com/wolfssl/wolfssl.git
cd wolfssl
./autogen.sh
./configure --enable-wolftpm --disable-examples --disable-crypttests --prefix="$PWD/../wolf_install"
make
make install
```

```
git clone https://github.com/wolfssl/wolftpm.git
cd wolftpm
./autogen.sh
./configure --enable-winapi --with-wolfcrypt="$PWD/../wolf_install" --prefix="$PWD/../wolf_install"
make
make install
```

These steps worked for me. Please let me know if you see any issues. 

Thanks,
David Garske, wolfSSL

ETHwsch, 2023-07-11:
I think I found the problem:

If I use "MSYS2 UCRT64" then I have the problems described above.
If I use "MSYS2 MSYS" it works as expected. I don't know exactly the differences between all the versions of MSYS1^2.

Question:

My use case is I have a VS2022 C++ Solution, 64 bit, Windows 11 and I want to use wolfTPM for any TPM related operations. Do I have to use "MSYS2 MSYS"  and use the Library from here (libwolfssl.dll.a, libwolftpm.dll.a):
```
$ ls -l
total 1104
-rw-r--r-- 1 root1 None 862868 Jul 11 09:48 libwolfssl.dll.a
-rw-r--r-- 1 root1 None    948 Jul 11 09:49 libwolfssl.la
-rw-r--r-- 1 root1 None 255508 Jul 11 09:55 libwolftpm.dll.a
-rw-r--r-- 1 root1 None   1050 Jul 11 09:55 libwolftpm.la
```
If so can I in VS, C++just  Load the Libs with LoadLibrary() and use the functions?

dgarske, 2023-07-11:
Hi @ETHwsch,

Have you tried our Visual Studio solution here? https://github.com/wolfSSL/wolfTPM/tree/master/wrapper
This is a C# wrapper, but it may also help you compile wolfTPM as a DLL you can use.

I am unsure the differences between the UCRT64 and normal MSYS. I'll see if I can investigate this.

Thanks,
David Garske, wolfSSL

ETHwsch, 2023-07-12:
I want to have a plain VS C++ based  unmanaged solution without any dependencies to .net. I found the following libs:

```
~/wolf_install/bin
-rwxr-xr-x 1 root1 None 752201 Jul 11 09:49 msys-wolfssl-40.dll
-rwxr-xr-x 1 root1 None 198875 Jul 11 09:55 msys-wolftpm-15.dll
```

If I use 'dumpbin.exe /exports', I can see all the TPM functions. 

```
File Type: DLL

  Section contains the following exports for msys-wolftpm-15.dll

    00000000 characteristics
    64AD0A8C time date stamp Tue Jul 11 09:53:48 2023
        0.00 version
           1 ordinal base
         372 number of functions
         372 number of names

    ordinal hint RVA      name

          1    0 000050F0 TPM2_ActivateCredential
          2    1 0000C780 TPM2_AppendPublic
          3    2 0001E5C0 TPM2_CalcCpHash
          4    3 0001E880 TPM2_CalcHmac
          5    4 0001E770 TPM2_CalcRpHash
....

```

This means it should be possible to use these DLLs. It seems that these DLLs have a dependency on 'msys-2.0.dll'. I am not familiar with MSYS at all, and I cannot estimate whether it is possible to link everything together. Has anybody ever done this before and are there instructions or examples?

dgarske, 2023-07-12:
Hi @ETHwsch ,

For the wolfCrypt library we have a very nice Visual Studio project in the root named `wolfssl.vcxproj`. You would need to update the IDE/WIN/user_settings.h with the one at examples/config/user_settings_wolftpm.h.

For wolfTPM we don't have a Visual Studio project yet. But you should be able to build a simple DLL using these instructions:
https://github.com/wolfSSL/wolfTPM/tree/master/wrapper/CSharp
I'll review adding a Visual Studio project for building the wolfTPM DLL.

I am not sure about msys either, but appears to supply a dependency DLL for the POSIX / STDLIB abstraction.

Are you able to continue your efforts with the DLL's you have now? Can you tell me more about your project and use for wolfTPM? Feel free to email support @ wolfssl.com directly if you would like a private discussion.

Thanks,
David Garske, wolfSSL