wolfSSL/wolfTPM #282: keygen -rsa -eh failed with TPM_RC_BAD_AUTH

TheBigFish, 2023-07-20:
I try to test wolftpm with ftpm (ms-tpm-20-ref)， while `keygen - rsa ` it works fine, but add parameter` -eh `, as `keygen - rsa -eh ` , get erros `Failure 0x9a2: TPM_RC_BAD_AUTH: Authorization failure without DA implications`:

```bash
TPM2.0 Key generation example
	Key Blob: keyblob.bin
	Algorithm: RSA
	Template: AIK
	Use Parameter Encryption: NULL
TPM2: Caps 0x00000000, Did 0x0000, Vid 0x0000, Rid 0x 0 
authCmd:
sessionHandle=0x40000009
nonceSize=0 nonceBuffer:
sessionAttributes=0x00
hmacSize=0 hmacBuffer:
Found 1 auth sessions
CommandProcess: Handles (Auth 1, In 1), CmdSz 355, AuthSz 9, ParamSz 328, EncSz 4
authCmd:
sessionHandle=0x40000009
nonceSize=0 nonceBuffer:
sessionAttributes=0x01
hmacSize=0 hmacBuffer:
D/TA:  TPM2_LINUX_SendCommand:106 TPM2_LINUX_SendCommand with pos 355, size 4096
D/TC:? 0 tee_ta_init_pseudo_ta_session:299 Lookup pseudo TA bc50d971-d4c9-42c4-82cb-343fb7f37896
D/TC:? 0 ldelf_load_ldelf:91 ldelf load address 0x40006000
D/LD:  ldelf:134 Loading TA bc50d971-d4c9-42c4-82cb-343fb7f37896
D/TC:? 0 ldelf_syscall_open_bin:140 Lookup user TA ELF bc50d971-d4c9-42c4-82cb-343fb7f37896 (Secure Storage TA)
D/TC:? 0 ldelf_syscall_open_bin:144 res=0xffff0008
D/TC:? 0 ldelf_syscall_open_bin:140 Lookup user TA ELF bc50d971-d4c9-42c4-82cb-343fb7f37896 (REE)
D/TC:? 0 ldelf_syscall_open_bin:144 res=0
D/LD:  ldelf:168 ELF (bc50d971-d4c9-42c4-82cb-343fb7f37896) at 0x4006d000
D/TA:  TA_CreateEntryPoint:129 Entry Point
D/TA:  _plat__NVEnable:377 _plat__NVEnable()
D/TA:  _plat__NvInitFromStorage:132 _plat__NvInitFromStorage()
I/TA: Created fTPM storage object, i: 0x0, s: 0x200, id: 0x54504d00, h:0x240cf0
I/TA: Created fTPM storage object, i: 0x1, s: 0x200, id: 0x54504d01, h:0x2408c0
I/TA: Created fTPM storage object, i: 0x2, s: 0x200, id: 0x54504d02, h:0x240660
I/TA: Created fTPM storage object, i: 0x3, s: 0x200, id: 0x54504d03, h:0x240400
I/TA: Created fTPM storage object, i: 0x4, s: 0x200, id: 0x54504d04, h:0x2401a0
I/TA: Created fTPM storage object, i: 0x5, s: 0x200, id: 0x54504d05, h:0x23ff40
I/TA: Created fTPM storage object, i: 0x6, s: 0x200, id: 0x54504d06, h:0x23fce0
I/TA: Created fTPM storage object, i: 0x7, s: 0x200, id: 0x54504d07, h:0x23fa80
I/TA: Created fTPM storage object, i: 0x8, s: 0x200, id: 0x54504d08, h:0x23f820
I/TA: Created fTPM storage object, i: 0x9, s: 0x200, id: 0x54504d09, h:0x23f5c0
I/TA: Created fTPM storage object, i: 0xa, s: 0x200, id: 0x54504d0a, h:0x23f360
I/TA: Created fTPM storage object, i: 0xb, s: 0x200, id: 0x54504d0b, h:0x23f100
I/TA: Created fTPM storage object, i: 0xc, s: 0x200, id: 0x54504d0c, h:0x23eea0
I/TA: Created fTPM storage object, i: 0xd, s: 0x200, id: 0x54504d0d, h:0x23ec40
I/TA: Created fTPM storage object, i: 0xe, s: 0x200, id: 0x54504d0e, h:0x23e9e0
I/TA: Created fTPM storage object, i: 0xf, s: 0x200, id: 0x54504d0f, h:0x23e780
I/TA: Created fTPM storage object, i: 0x10, s: 0x200, id: 0x54504d10, h:0x23e520
I/TA: Created fTPM storage object, i: 0x11, s: 0x200, id: 0x54504d11, h:0x23e2c0
I/TA: Created fTPM storage object, i: 0x12, s: 0x200, id: 0x54504d12, h:0x23e060
I/TA: Created fTPM storage object, i: 0x13, s: 0x200, id: 0x54504d13, h:0x23de00
I/TA: Created fTPM storage object, i: 0x14, s: 0x200, id: 0x54504d14, h:0x23dba0
I/TA: Created fTPM storage object, i: 0x15, s: 0x200, id: 0x54504d15, h:0x23d940
I/TA: Created fTPM storage object, i: 0x16, s: 0x200, id: 0x54504d16, h:0x23d6e0
I/TA: Created fTPM storage object, i: 0x17, s: 0x200, id: 0x54504d17, h:0x23d480
I/TA: Created fTPM storage object, i: 0x18, s: 0x200, id: 0x54504d18, h:0x23d220
I/TA: Created fTPM storage object, i: 0x19, s: 0x200, id: 0x54504d19, h:0x23cfc0
I/TA: Created fTPM storage object, i: 0x1a, s: 0x200, id: 0x54504d1a, h:0x23cd60
I/TA: Created fTPM storage object, i: 0x1b, s: 0x200, id: 0x54504d1b, h:0x23cb00
I/TA: Created fTPM storage object, i: 0x1c, s: 0x200, id: 0x54504d1c, h:0x23c8a0
I/TA: Created fTPM storage object, i: 0x1d, s: 0x200, id: 0x54504d1d, h:0x23c640
I/TA: Created fTPM storage object, i: 0x1e, s: 0x200, id: 0x54504d1e, h:0x23c3e0
I/TA: Created fTPM storage object, i: 0x1f, s: 0x200, id: 0x54504d1f, h:0x23c180
I/TA: Created fTPM storage object, i: 0x20, s: 0x200, id: 0x54504d20, h:0x23bf20
D/TA:  TA_CreateEntryPoint:151 NVEnable Complete
D/TA:  TA_CreateEntryPoint:158 TPM_Manufacture
D/TA:  _plat__NvWriteBack:288 bMap: 0xffffffff
D/TA:  _plat__NVEnable:377 _plat__NVEnable()
D/TA:  TA_CreateEntryPoint:170 Init Complete
D/TA:  TA_CreateEntryPoint:193 No TPM state present
D/TA:  _plat__NvWriteBack:288 bMap: 0x6
D/TC:? 0 tee_ta_init_pseudo_ta_session:299 Lookup pseudo TA 3a2f8978-5dc0-11e8-9c2d-fa7ae01bbebc
D/TC:? 0 tee_ta_init_pseudo_ta_session:312 Open system.pta
D/TC:? 0 tee_ta_init_pseudo_ta_session:329 system.pta : 3a2f8978-5dc0-11e8-9c2d-fa7ae01bbebc
D/TA:  fTPM_Submit_Command:342 Success, RS: 0x21a
D/TC:? 0 tee_ta_close_session:512 csess 0x37507370 id 1
D/TC:? 0 tee_ta_close_session:531 Destroy session
ResponseProcess: Handles (Out 1), RespSz 538, ParamSz 515, DecSz 314, AuthSz 5
TPM2_CreatePrimary: 0x80000000 (314 bytes)
TPM2_GetNonce (32 bytes)
D/TA:  TPM2_LINUX_SendCommand:106 TPM2_LINUX_SendCommand with pos 59, size 4096
D/TC:? 0 tee_ta_init_session_with_context:607 Re-open TA bc50d971-d4c9-42c4-82cb-343fb7f37896
D/TA:  _plat__NvWriteBack:288 bMap: 0x100
D/TA:  fTPM_Submit_Command:342 Success, RS: 0x30
D/TC:? 0 tee_ta_close_session:512 csess 0x374fbad0 id 3
D/TC:? 0 tee_ta_close_session:531 Destroy session
Session Key 0
TPM2_StartAuthSession: handle 0x3000000, algorithm NULL
TPM2_StartAuthSession: sessionHandle 0x3000000
authCmd:
sessionHandle=0x40000009
nonceSize=0 nonceBuffer:
sessionAttributes=0x00
hmacSize=0 hmacBuffer:
Found 1 auth sessions
CommandProcess: Handles (Auth 1, In 2), CmdSz 41, AuthSz 9, ParamSz 10, EncSz 0
authCmd:
sessionHandle=0x40000009
nonceSize=0 nonceBuffer:
sessionAttributes=0x01
hmacSize=0 hmacBuffer:
D/TA:  TPM2_LINUX_SendCommand:106 TPM2_LINUX_SendCommand with pos 41, size 4096
D/TC:? 0 tee_ta_init_session_with_context:607 Re-open TA bc50d971-d4c9-42c4-82cb-343fb7f37896
D/TA:  fTPM_Submit_Command:342 Success, RS: 0x1d
D/TC:? 0 tee_ta_close_session:512 csess 0x374fb390 id 3
D/TC:? 0 tee_ta_close_session:531 Destroy session
ResponseProcess: Handles (Out 0), RespSz 29, ParamSz 10, DecSz 0, AuthSz 5
policySecret applied on session
RSA AIK template
Creating new RSA key...
authCmd:
sessionHandle=0x03000000
nonceSize=32 nonceBuffer:
	00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
	00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
sessionAttributes=0x00
hmacSize=0 hmacBuffer:
Found 1 auth sessions
CommandProcess: Handles (Auth 1, In 1), CmdSz 370, AuthSz 41, ParamSz 311, EncSz 21
TPM2_GetNonce (32 bytes)
Name 0: 0
cpHash: cmd 153, size 32
	7c 15 94 5f 4a 43 c1 a4 6f 8c 7b 0a 28 e7 24 29 | |.._JC..o.{.(.$)
	a3 d0 05 f1 a4 e8 ea 47 f0 ab 13 5c 22 95 4f 44 | .......G...\".OD
HMAC Key: 0
HMAC Auth: attrib 0, size 32
	85 d4 1a 14 43 d5 41 54 b7 0c 4b 40 56 7e 60 e6 | ....C.AT..K@V~`.
	9b b7 3b d6 2e 42 df 35 5d 46 5d b3 73 e3 d9 60 | ..;..B.5]F].s..`
authCmd:
sessionHandle=0x03000000
nonceSize=32 nonceBuffer:
	d3 36 d0 d5 ad 62 fb 39 57 db 06 81 81 28 0b f7 | .6...b.9W....(..
	a3 cd b6 73 d5 bd c5 76 18 4f 06 c5 59 4d a2 8e | ...s...v.O..YM..
sessionAttributes=0x00
hmacSize=32 hmacBuffer:
	85 d4 1a 14 43 d5 41 54 b7 0c 4b 40 56 7e 60 e6 | ....C.AT..K@V~`.
	9b b7 3b d6 2e 42 df 35 5d 46 5d b3 73 e3 d9 60 | ..;..B.5]F].s..`
D/TA:  TPM2_LINUX_SendCommand:106 TPM2_LINUX_SendCommand with pos 370, size 4096
D/TC:? 0 tee_ta_init_session_with_context:607 Re-open TA bc50d971-d4c9-42c4-82cb-343fb7f37896
D/TA:  fTPM_Submit_Command:342 Success, RS: 0xa
D/TC:? 0 tee_ta_close_session:512 csess 0x374fb390 id 3
D/TC:? 0 tee_ta_close_session:531 Destroy session
TPM2_Create key failed 2466: TPM_RC_BAD_AUTH: Authorization failure without DA implications
wolfTPM2_CreateKey failed

Failure 0x9a2: TPM_RC_BAD_AUTH: Authorization failure without DA implications

D/TA:  TPM2_LINUX_SendCommand:106 TPM2_LINUX_SendCommand with pos 14, size 4096
D/TC:? 0 tee_ta_init_session_with_context:607 Re-open TA bc50d971-d4c9-42c4-82cb-343fb7f37896
D/TA:  fTPM_Submit_Command:342 Success, RS: 0xa
D/TC:? 0 tee_ta_close_session:512 csess 0x374fb390 id 3
D/TC:? 0 tee_ta_close_session:531 Destroy session
TPM2_FlushContext: Closed handle 0x80000000
D/TA:  TPM2_LINUX_SendCommand:106 TPM2_LINUX_SendCommand with pos 12, size 4096
D/TC:? 0 tee_ta_init_session_with_context:607 Re-open TA bc50d971-d4c9-42c4-82cb-343fb7f37896
D/TA:  _plat__NvWriteBack:288 bMap: 0x306
D/TA:  fTPM_Submit_Command:342 Success, RS: 0xa
D/TC:? 0 tee_ta_close_session:512 csess 0x374fb390 id 3
D/TC:? 0 tee_ta_close_session:531 Destroy session
tpcm/basic/wolftpm/test_wolftpm.c:122::FAIL: Expected 0x00000000 Was 0x000009A2
```
Thanks!

dgarske, 2023-07-20:
Hi @TheBigFish ,

Thank you for the report. I can reproduce and will investigate. Seems to just be an issue with the endorsement hierarchy .

Can you tell me more about your TPM use case?

```
% ./examples/keygen/keygen -rsa -eh
TPM2.0 Key generation example
	Key Blob: keyblob.bin
	Algorithm: RSA
	Template: AIK
	Use Parameter Encryption: NULL
wolfSSL Entering wolfCrypt_Init
TPM2: Caps 0x00000000, Did 0x0000, Vid 0x0000, Rid 0x 0
TPM2_Startup pass
TPM2_SelfTest pass
TPM2_CreatePrimary: 0x80000000 (314 bytes)
TPM2_StartAuthSession: handle 0x3000002, algorithm NULL
TPM2_StartAuthSession: sessionHandle 0x3000002
policySecret applied on session
RSA AIK template
Creating new RSA key...
TPM2_Create key failed 2466: TPM_RC_BAD_AUTH: Authorization failure without DA implications
wolfTPM2_CreateKey failed

Failure 0x9a2: TPM_RC_BAD_AUTH: Authorization failure without DA implications

TPM2_FlushContext: Closed handle 0x80000000
wolfSSL Entering wolfCrypt_Cleanup
```

TheBigFish, 2023-07-21:
I use the ftpm as a DTA in optee_os as TA1，and use wolftpm and wolfssl in another DTA as TA2。
In TA2, I call keygen with "-rsa -eh"。
Does this information enough?

dgarske, 2023-07-24:
Hi @TheBigFish ,

The `keygen -rsa -eh` successfully creates a primary handle for the EH, however it fails with an auth error when trying to create a child RSA key. I've tried to find some examples of this working with tpm2_tools, but not seeing any. Do you have some working examples of using the EH you can share? 

What are you planning to use the child EH key for? Is this a make/activate credential? If so you might have been luck with `./examples/attestation/make_credential` and `./examples/attestation/activate_credential`.

Thanks,
David Garske, wolfSSL

TheBigFish, 2023-07-31:
Hi @dgarske 
I just check the README in examples/attestation, and get this: 
> Note: All of these example allow the use of the Endorsement Key and Attestation Key under the Endorsement Hierarchy. This is done by adding the `-eh` option when executing any of the three examples above. 

The first step in "Example usage" is `$ ./examples/keygen/keygen -rsa`,  so I add `-eh` to test it ,then get an error.

Thanks!

dgarske, 2023-08-01:
Hi @TheBigFish ,

Thank you for pointing that out. I'll continue to investigate.

Thanks,
David Garske, wolfSSL

dgarske, 2023-09-01:
FYI: This issue is now documented in our CI testing here: https://github.com/wolfSSL/wolfTPM/blob/master/examples/run_examples.sh#L105
I plan to work on resolving this next week. It seems the policy secret code around the use of -eh is broken.

dgarske, 2023-12-29:
Hi @TheBigFish ,

I finally made time to resolve the endorsement key issues. The problem was very minor with a change to not use created loaded and also an issue with not populating the "name" field correctly. Fixes have been pushed to: https://github.com/wolfSSL/wolfTPM/pull/320

Thanks,
David Garske, wolfSSL

closed 2023-12-29 by embhorn