wolfSSL/wolfTPM #349: Is 1,000,000 a reasonable value for TPM_TIMEOUT_TRIES?

gojimmypi, 2024-04-30:
See [wolftpm/tpm2_types.h#L388](https://github.com/wolfSSL/wolfTPM/blob/9d03c890fee5e637f58c3d63132acb8196e254fd/wolftpm/tpm2_types.h#L388) for `#define TPM_TIMEOUT_TRIES 1000000`.

Perhaps a few orders of magnitude smaller is more reasonable? Even if after, say 10 attempts, should the I2C operation [keep trying](https://github.com/wolfSSL/wolfTPM/blob/9d03c890fee5e637f58c3d63132acb8196e254fd/src/tpm2_tis.c#L267)?

This might be a good place for an optional user-specified timeout, too.

dgarske, 2024-04-30:
It is an overridable macro. For slower microcontrollers a value like 1,000 or 10,000 may be more appropriate. To avoid a busy spin you can also implement XTPM_WAIT() to enable some sleep() / context switching.

gojimmypi, 2024-04-30:
ok, I'll leave it.

closed 2024-04-30 by gojimmypi