wolfSSL/wolfTPM #356: P-256K1 support in the TCG standards

jarkkojs, 2024-05-29:
Not exactly wolfTPM issue/bug per se, but more like querying on topic. 

I recently have implemented a patch set for asymmetric TPM2 RSA/ECDSA keys [1], and that made me consciously realize that P256K1 is not part of TCG Algorithm Registry. Any ideas why it has never proceeded to the registry? 

[1] https://lore.kernel.org/linux-crypto/20240528210823.28798-1-jarkko@kernel.org/

jarkkojs, 2024-05-29:
As a workaround I'm going to do this next after tpm2_key_*: https://social.kernel.org/notice/AiNuw35YY9uOSrhiK0

dgarske, 2024-05-29:
Hi @jarkkojs ,

The NIST Prime curves like SECP256R1 are the most popular. The Koblitz curves are less common and the TPM manufactures are not required to support that curve. What is prompting your interest in using the Koblitz ECC curves?

@kgoldman any feedback on the status of TCG and the Koblitz curves like SECP256K1?

Thanks,
David Garske, wolfSSL

kgoldman, 2024-05-30:
The TCG process is to include an algorithm when there is demand. The demand can come from any TCG member company, but often comes through one of the work groups.

jarkkojs, 2024-05-30:
@kgoldman, @dgarske: So, first of all thank you for these insights. This verified what I assumed how it was.

Secondly, I put together TPM2 backed ECDSA keys for NIST-{P256,P384,P521} for Linux kernel keyring, of which I'm co-maintainer of together with David Howells from Red Hat. They use TPM2 for signing and software ECDSA for verification using pre-existing implementation from Stefan Berger (IBM). Apparently this was also first signing implementation across the board in Linux, as Stefan's implementation does only verification.

It is working pretty well, I've tested it with keys created by both IBM and Intel TPM2 stacks, and I'm also working on a tool called "tpm2ctl", which is really not yet another TPM2 stack. It will just a few sub-commands to interact with both keyring and TPM2 device to quickly get x.509 compatible key pair ready and set. It also takes away some openssl commands with the help of libsodium.

All of this work is just implementing TPM2 parts of this spec authored by David Woodhouse, who is principal engineer from AWS: https://datatracker.ietf.org/doc/draft-woodhouse-cert-best-practice/

I'm the only person in this equation who has no corporate affiliation or any other org affiliation, except perhaps kernel.org ;-) Had a good and also legit time to implement the feature given zero affiliations. I've been a contract researcher since Mar 2023 in https://www.sochub.fi/, which is ending at the end of Sep, and I'm returning to the corporate world, so time was running up to do this as an actor with no interest group, except my own interest towards tech :-)

So. This brings me to my point. The feature right now with NIST stuff helps companies like Amazon to ease working with x.509 certificates e.g. in data centers. However, p256k1 is vastly popular in open source communities such as Bitcoin and Ethereum and some other blockchain technologies. I would feel that it would be balanced if also blockchains would get the benefits of securing the private keys of crypto wallets and smart contracts (Eth).

From kernel side, given the work I already did, this would not a big stretch to enable also that ECDSA. From TCG's side it would require putting that curve also to the registry. It's not like either that P256K1 would be non-existent in proprietary world. E.g. Intel CSE natively supports it.

Right now there is a market of USB cryptowallets that only exist IMHO because TPM2 chip is not delivering here. IMHO, everyone would be better off, if TPM2 had also this feature. Then it would pretty well encapsulate full coverage of crypto ecosystem of 2024.

jarkkojs, 2024-05-30:
I always keep my integrity in any possible situation, no matter which corporate I'm hired or not but just thought that this is the most legit position for me to propose anything in curve space. I'm only doing charity work and enabling tech... for free of charge ;-) On my leisure time...

kgoldman, 2024-05-30:
You might already know this, but TCG has a provision for someone to be an 'invited expert'. You can join calls, contribute to documents, everything except vote. I'm sure you would qualify.

jarkkojs, 2024-05-31:
@kgoldman I'll definitely consider that! Thank you.

closed 2024-08-22 by dgarske