wolfSSL/wolfTPM #358: Support of tpmrmX interface

IldarAbdullin-okta, 2024-06-07:
Hi,

Are there any plans to support tpm resource manager integration(tpmrm0)?

Ildar

dgarske, 2024-06-08:
Hi @IldarAbdullin-okta ,

No plans currently. What is the use-case and platform where you need this feature? 

If you are on Linux we support `--enable-tislock` or `WOLFTPM_TIS_LOCK` that uses a named semaphore for locking allowing concurrent access between processes. However that only works if using our TIS layer. If using the /dev/tpm0 it's handled by the Linux Kernel.

Thanks,
David Garske, wolfSSL

IldarAbdullin-okta, 2024-06-08:
> Hi @IldarAbdullin-okta ,
>
> No plans currently. What is the use-case and platform where you need this feature?
>
> If you are on Linux we support `--enable-tislock` or `WOLFTPM_TIS_LOCK` that uses a [...]

Platform is Linux, and yes, we are using /dev/tpm0 interface. One of the concern with the approach is that communication with /dev/tpmX requires root privileges. As a workaround we are thinking about installing udev rule similar to - [tpm2-tss tool approach](https://github.com/tpm2-software/tpm2-tss/blob/master/dist/tpm-udev.rules) and add an end user to TSS group.

As a workaround we can have `KERNEL=="tpm[0-9]*", TAG+="systemd", MODE="0660", GROUP="tss"`, and this most likely will be our short term solution. However that rule can be overwritten if user will install tpm2-tss tool after our application.  So for us it would be great if we can follow the same approach as TSS has today for their tpm tools

dgarske, 2024-06-20:
Hi @IldarAbdullin-okta ,

Thank you for those details. I will look into this TSS approach and see if I can improve wolfTPM support.

Thanks,
David Garske, wolfSSL

dgarske, 2024-07-31:
Hi @IldarAbdullin-okta , I posted instructions for setting up a custom group and Udev rules. See https://github.com/wolfSSL/wolfTPM/pull/366/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R273

IldarAbdullin-okta, 2024-08-01:
> Hi @IldarAbdullin-okta , I posted instructions for setting up a custom group and Udev rules. See https://github.com/wolfSSL/wolfTPM/pull/366/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R273

Hi @dgarske ,

Thanks for updating instructions! New changes look good to me. One thing is that `sudo udevadm control -R` doesn't help and you still have to logout or reboot(tested on Ubuntu 22+). I didn't find a good way how to make it work without reboot. Most likely that requires reload of a TPM driver

Also checking regarding tpmrmX integration again. I'm hearing feedback that integration against tpmrmX may help against apps that are trying to DoS TPM with a lot of requests.

Ildar

dgarske, 2024-08-01:
> Hi @dgarske ,
>
> Thanks for updating instructions! New changes look good to me. One thing is that `sudo udevadm control -R` doesn't help and you still have to logout or reboot(tested on Ubuntu 22+). I [...]

I noticed that too about having to reboot. I will update the instructions.

PR #366 (2024-08-01): Various improvements for `/dev/tpm0`
* Fix for /dev/tpm0 file descriptor check (fixes #365). 
* Added documentation for /dev/tpm0 permissions (fixes #358). 
* Various spelling and documentation cleanups.
commit 6951b8d3 (2024-08-01): Fix for /dev/tpm0 file descriptor check (fixes #365). Added documentation for /dev/tpm0 permissions (fixes #358). Various spelling and documentation cleanups.
closed 2024-08-01 by embhorn