wolfSSL/wolfTPM #365: tpm2_linux open() file descriptor 0 should be allowed

radomik, 2024-07-31:
Hi,
I'm using older version of WolfTPM but see this part of code is same in newest version.

What my program does is intense usage of TPM2_GetRandom from few threads in parallel, for couple of minutes, that's needed for TLS encryption.

Observation is that in some cases /dev/tpm0 cannot be opened with `errno=Success` that caused then cascade of `errno=Device or resource busy`.

As from Linux man for open(0 can be seen that open() valid return value is nonnegative integer, thus 0 is valid value.

Suggest to change a condition fro `>` to `>=` to:
```c
    fd = open(TPM2_LINUX_DEV, O_RDWR | O_NONBLOCK);
    if (fd >= 0) {
        /* Send the TPM command */
```

After fix, I see no more issues with my code using wolfTPM library.

dgarske, 2024-07-31:
Hi @radomik , thank you so much for the detailed report! I'll put up a fix shortly and reference it here.
Thanks, David Garske, wolfSSL

dgarske, 2024-07-31:
Hi @radomik , I posted the fix in PR #366.

PR #366 (2024-08-01): Various improvements for `/dev/tpm0`
* Fix for /dev/tpm0 file descriptor check (fixes #365). 
* Added documentation for /dev/tpm0 permissions (fixes #358). 
* Various spelling and documentation cleanups.
commit 6951b8d3 (2024-08-01): Fix for /dev/tpm0 file descriptor check (fixes #365). Added documentation for /dev/tpm0 permissions (fixes #358). Various spelling and documentation cleanups.
closed 2024-08-01 by embhorn