wolfSSL/wolfTPM #370: Need help - can't decrypt data using wolfTPM2_RsaDecrypt

Eznopot, 2024-08-15:
Hi, I'm a beginner with TPM and I work on a project where I need to use the TPM to encrypt/decrypt some data.
Using the example I have created 3 function:
- generateKey: generate RSA Key and storing on the TPM
- Encrypt: Read the tpm to get RSA key and encrypt data using the function `wolfTPM2_RsaEncrypt`
- Decrypt: Read the tpm to get RSA Key and decrypt data using the function `wolfTPM2_RsaDecrypt`

The 2 first one seems working, I dont have any error but when I try to decrypt data I have this error: `TPM2_RSA_Decrypt failed 2466: TPM_RC_BAD_AUTH: Authorization failure without DA implications`.
What's that's mean ? Do I need to set special attribute or something to my key ?
Thanks for your help !

Note: TPM2.0 chip is slb9670 and my function looks like this:
```C
ResultDecrypt *decrypt(byte *msg, int len)
{
    void *userCtx = NULL;
    int rc;
    WOLFTPM2_DEV dev;
    WOLFTPM2_KEY storage;
    WOLFTPM2_KEYBLOB keyBlob;
    WOLFTPM2_SESSION tpmSession;
    WOLFTPM2_HANDLE parent;
    WOLFTPM2_NV nv;
    TPM2B_AUTH auth;
    word32 readSize;
    TPMI_RH_NV_AUTH authHandle = TPM_RH_OWNER;
    int paramEncAlg = TPM_ALG_NULL;
    int offset = 0;
    /* Needed for TPM2_ParsePublic */
    byte pubAreaBuffer[sizeof(TPM2B_PUBLIC)];
    int pubAreaSize;
    word32 nvIndex = TPM2_DEMO_NVRAM_STORE_INDEX;

    printf("NV Read\n");
    printf("\tNV Index: 0x%08x\n", nvIndex);
    printf("\tAuth: %s\n",
           (authHandle == TPM_RH_ENDORSEMENT) ? "Endorsement" : (authHandle == TPM_RH_PLATFORM) ? "Platform"
                                                                                                : "Owner");

    XMEMSET(&keyBlob, 0, sizeof(keyBlob));
    XMEMSET(&tpmSession, 0, sizeof(tpmSession));
    XMEMSET(&parent, 0, sizeof(parent));
    XMEMSET(&auth, 0, sizeof(auth));
    XMEMSET(&storage, 0, sizeof(storage));

    rc = wolfTPM2_Init(&dev, TPM2_IoCb, userCtx);
    if (rc != TPM_RC_SUCCESS)
    {
        printf("\nwolfTPM2_Init failed\n");
        goto exit2;
    }

    /* get SRK */
    rc = getPrimaryStoragekey(&dev, &storage, TPM_ALG_RSA);
    if (rc != 0)
        goto exit2;

    auth.size = sizeof(gKeyAuth) - 1;
    XMEMCPY(auth.buffer, gKeyAuth, auth.size);

    /* Prepare auth for NV Index */
    XMEMSET(&nv, 0, sizeof(nv));
    nv.handle.hndl = nvIndex;
    nv.handle.auth.size = auth.size;
    XMEMCPY(nv.handle.auth.buffer, auth.buffer, auth.size);

    readSize = sizeof(keyBlob.pub.size);
    printf("Trying to read %d bytes of public key size marker\n", readSize);
    rc = wolfTPM2_NVReadAuth(&dev, &nv, nvIndex,
                             (byte *)&keyBlob.pub.size, &readSize, 0);
    if (rc != 0)
    {
        printf("Was a public key part written? (see nvram/store)\n");
        goto exit2;
    }
    printf("Successfully read public key part from NV\n\n");
    offset += readSize;

    readSize = sizeof(UINT16) + keyBlob.pub.size; /* account for TPM2B size marker */
    printf("Trying to read %d bytes of public key part from NV\n", keyBlob.pub.size);
    rc = wolfTPM2_NVReadAuth(&dev, &nv, nvIndex,
                             pubAreaBuffer, &readSize, offset);
    if (rc != 0)
        goto exit2;
    printf("Successfully read public key part from NV\n\n");
    offset += readSize;

    /* Necessary for storing the publicArea with the correct encoding */
    rc = TPM2_ParsePublic(&keyBlob.pub, pubAreaBuffer,
                          (word32)sizeof(pubAreaBuffer), &pubAreaSize);
    if (rc != TPM_RC_SUCCESS)
    {
        printf("Decoding of PublicArea failed. Unable to extract correctly.\n");
        goto exit2;
    }

    printf("Trying to read size marker of the private key part from NV\n");
    readSize = sizeof(keyBlob.priv.size);
    rc = wolfTPM2_NVReadAuth(&dev, &nv, nvIndex,
                             (byte *)&keyBlob.priv.size, &readSize, offset);
    if (rc != 0)
    {
        printf("Was a private key part written? (see nvram/store)\n");
        goto exit2;
    }
    printf("Successfully read size marker from NV\n\n");
    offset += readSize;

    readSize = keyBlob.priv.size;
    printf("Trying to read %d bytes of private key part from NV\n", readSize);
    rc = wolfTPM2_NVReadAuth(&dev, &nv, nvIndex,
                             (byte *)&keyBlob.priv.buffer, &readSize, offset);
    if (rc != 0)
        goto exit2;
    printf("Successfully read private key part from NV\n\n");

    /* auth 0 is owner, no auth */
    wolfTPM2_SetAuthPassword(&dev, 0, &auth);
    wolfTPM2_UnsetAuth(&dev, 1);

    parent.hndl = authHandle;
    //! this remove the Auth and the key: rc = wolfTPM2_NVDeleteAuth(&dev, &parent, nvIndex);
    if (rc != 0)
        goto exit2;

    printf("Extraction of key from NVRAM at index 0x%x succeeded\n",
           nvIndex);

    printf("Trying to load the key extracted from NVRAM\n");

    rc = wolfTPM2_LoadKey(&dev, &keyBlob, &storage.handle);
    if (rc != TPM_RC_SUCCESS)
    {
        printf("wolfTPM2_LoadKey failed\n");
        goto exit2;
    }
    printf("Loaded key to 0x%x\n",
           (word32)keyBlob.handle.hndl);

exit2:
    if (rc != 0)
    {
        printf("\nFailure 0x%x: %s\n\n", rc, wolfTPM2_GetRCString(rc));
        return NULL;
    }
    int msgSize = len;
    rc = wolfTPM2_RsaDecrypt(&dev, &keyBlob, paramEncAlg, msg, msgSize, resultDecrypt.Res, &resultDecrypt.ResSize);
    if (rc != 0)
    {
        printf("\nFailure 0x%x: %s\n\n", rc, wolfTPM2_GetRCString(rc));
    }
    wolfTPM2_UnloadHandle(&dev, &keyBlob.handle);
    wolfTPM2_UnloadHandle(&dev, &tpmSession.handle);
    wolfTPM2_Cleanup(&dev);
    return &resultDecrypt;
}
```
(the code is pretty bad because I didn't clean it yet sorry😓)

dgarske, 2024-08-15:
Hi @Eznopot ,

You did not show how you created the RSA key, but likely you used a password for authentication. 

For this to work you need to set the password via keyBlob.handle.auth.size and keyBlob.handle.auth.buffer before calling wolfTPM2_RsaDecrypt. The wrappers calls wolfTPM2_SetAuthHandle, so it will override your attempt to do this and will set a blank password.

Thanks,
David Garske, wolfSSL

Eznopot, 2024-08-16:
Hello,
Thanks for your answer!
To create the RSA key, I use the same code as in `examples/keygen/keygen.c`.
I'll try to put the authentication in the key!
I'll send you another message if it works.

Eznopot, 2024-08-16:
It works !
Thanks you so much !
I leave my star on the repo !
Have a nice day.

closed 2024-08-16 by Eznopot